OpenframeProject/Openframe #3: Why an account?

taseenb, 2016-03-08:
Hi. Congratulations for the project, i just installed it on a Raspberry Pi and think the whole idea sounds quite cool, but i don't understand the need for an account (mandatory). This looks like a serious compromise for an open source project. What is this account supposed to be used for? Thanks

jmwohl, 2016-03-08:
Hey @taseenb — thanks for the question, happy to clarify. 

Openframe consists of several pieces — one is this repo, which is the software that runs on the frame and controls the playing of artworks. The other main component is the [server](https://github.com/OpenframeProject/Openframe-APIServer), which is the central location for storing data about frames, users, and artworks. Your account lets you add and share artwork and push it to your frame remotely. We'll be continuing to add functionality, and hope that others will contribute in the form of extensions.

We're hosting an instance of the server at openframe.io, but the entire project is open source; there's no reason why you couldn't host your own private Openframe server as well (you'd need an account on your own server in that case... user accounts are central to the platform's data model). Over the coming weeks we'll be adding to the documentation about this.

taseenb, 2016-03-08:
Hey @jmwohl thanks for the explanation!

I understand that one could make her/his own server, but i am wondering if the server side could be optional, to give more flexibility for quick setup and distribution. Is that out of the scope of your idea?

jmwohl, 2016-03-08:
No problem — for the present, making the server side optional is out of scope. The frame controller is tied pretty tightly to the idea that it will be managed remotely so that A) users don't need to muck around on the frame every time they want to change artwork, and B) users can discover and push artwork from a community of users.

That said, you're welcome to fork this repo and see about de-coupling it from the server component. Not sure how helpful it would be over simply executing commands to fire up artwork directly on the pi.

closed 2016-03-20 by jmwohl