OpenframeProject/Openframe #5: Error when pushing shader artwork

chicken-bit, 2016-03-30:
When attempting to push any shader, receive the following error message:

glslViewer: src/app.cpp:250: void initGL(int, char **): Assertion 'surface != ((EGLSurface)0)' failed.

Running on a Rasp Pi Model B (hardware revision 000e), linux 4.1.19

Images seem to work just fine, only currently happening with any shaders I try

jmwohl, 2016-03-31:
Hmm... I've seen this error in the past, but I think a reboot fixed it. I'm not the shader expert, maybe @patriciogonzalezvivo will have an idea? Will see if he has any input...

patriciogonzalezvivo, 2016-03-31:
Hello @chicken-bit , seams like you are running an old version of glslViewer. Could you try re installing it glslViewer? For it follow the directions https://github.com/patriciogonzalezvivo/glslViewer#installing-in-raspberrypi

chicken-bit, 2016-04-01:
Hi @patriciogonzalezvivo , no dice on reinstalling glslViewer, still receiving the same error

patriciogonzalezvivo, 2016-04-01:
Interesting... are you running it as 'pi' user? If not, do your user have privileges on the group 'video'??

chicken-bit, 2016-04-04:
Yup, running as pi user and also member of video group
pi@openframe:~ $ id
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),101(input),108(netdev),997(gpio),998(i2c),999(spi)

indefinit, 2016-04-11:
I'm seeing the same behavior on a Pi2 running the same OS as @chicken-bit but the problem is also occurring on images for me as well.

patriciogonzalezvivo, 2016-04-11:
Can you send the console error?

indefinit, 2016-04-11:
just rebooted and tried to replicate the error message and now it just hangs with a blinking cursor, no visible error message.

indefinit, 2016-04-11:
ah wait, I tried recompiling glslViewer and now both shaders and images seem to be working as expected.  Sorry about that!

jmwohl, 2016-04-11:
Hmm, still a little strange, shouldn't need to compile glslViewer again... I'll take a look at the glslViewer install that is done by the glslViewer extension and make sure it's up to date.

jmwohl, 2016-04-11:
I haven't been able to replicate this, but I did release a new version of openframe this morning that might help... in the previous version, the glslViewer extension was getting installed twice, once as a dep of openframe package and once as a global dep. The new release has some other minor improvements too (artworks switch much faster, e.g.).

@indefinit - just for reference, you can get debug output to display by setting the DEBUG env var. For example running `DEBUG=* openframe` instead of just `openframe` will display the full debug output. It's probably mostly meaningless to anyone but me at the moment :), but it might help diagnose problems.

jmwohl, 2016-09-24:
I believe this is resolved, closing.

closed 2016-09-24 by jmwohl