OpenframeProject/Openframe #7: Does Openframe work on Raspbian Wheezy or Jessie?

indefinit, 2016-04-09:
I seem to be getting a bunch of errors when trying to install Openframe on Wheezy with Pi2.  At first I thought maybe the problem was related to me running gcc-4.6.3 but now I'm thinking maybe it has to do with the operating system in general.  I'm able to go through all of the installation steps except the script fails when I get to mmmagic@0.4.1 install: `node-gyp rebuild`.  Any help is much appreciated!  Thanks!

Error in console says:
``` 
npm ERR! Failed at the mmmagic@0.4.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the mmmagic package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mmmagic
npm ERR! There is likely additional logging output above.
```

jmwohl, 2016-04-09:
We've been working off of the latest NOOBS, which installs Jessie... that's definitely the most sure-fire way to get up and running with Openframe (though it does mean flashing a new SD card). That said, I had openframe running on Wheezy several months ago (before the install script was available), so I think it should work...

Are you using the [install script](http://openframe.io/install.sh), or installing openframe directly via NPM? The first thing I'd check is that you have your [NPM perms set correctly](https://docs.npmjs.com/getting-started/fixing-npm-permissions).

I'll also take a look and see where in the dep tree mmmagic is being required. It would be great to get this working smoothly on Wheezy as well as Jessie.

indefinit, 2016-04-09:
At first I was using the install script which attempted to install openframe under nvm.  After the script failed install because of node-gyp / mmagic, I then tried uninstalling nvm, and reinstalling the latest node + the global npm install of openframe.  npm install seems to fail in the same place.  Log output:

```
8001 error Linux 3.18.5-v7+
8002 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "openframe"
8003 error node v0.12.6
8004 error npm  v2.11.2
8005 error code ELIFECYCLE
8006 error mmmagic@0.4.1 install: `node-gyp rebuild`
8006 error Exit status 1
8007 error Failed at the mmmagic@0.4.1 install script 'node-gyp rebuild'.
8007 error This is most likely a problem with the mmmagic package,
8007 error not with npm itself.
8007 error Tell the author that this fails on your system:
8007 error     node-gyp rebuild
8007 error You can get their info via:
8007 error     npm owner ls mmmagic
8007 error There is likely additional logging output above.
8008 verbose exit [ 1, true ]
8009 verbose unbuild lib/node_modules/openframe/node_modules/http-request/node_modules/mmmagic
8010 info preuninstall mmmagic@0.4.1
8011 info uninstall mmmagic@0.4.1
8012 verbose unbuild rmStuff mmmagic@0.4.1 from /usr/local/lib/node_modules
8013 verbose unbuild rmStuff in /usr/local/lib/node_modules/openframe/node_modules/http-request/node_modules
8014 info postuninstall mmmagic@0.4.1
```

jmwohl, 2016-04-09:
Ok – it might be related to the gcc version available in Wheezy. I'd try upgrading to gcc 4.8... something like this:

https://somewideopenspace.wordpress.com/2014/02/28/gcc-4-8-on-raspberry-pi-wheezy/

If you have a chance to try this, great... if not, I'll take a stab at it as soon as I can.

jmwohl, 2016-04-09:
Oh wait, sorry — you're attempting to install it with node v0.12.6. I'd definitely [upgrade to node 4.x LTS](http://blog.wia.io/installing-node-js-v4-0-0-on-a-raspberry-pi), that might be all it takes.

indefinit, 2016-04-10:
I think I went down a rabbit hole on this one.  So I can confirm that npm install -g openframe seems to need both gcc 4.8 and node 4.x, but I wasn't successful getting gcc to upgrade to 4.8 on wheezy.  I also tried using node 4.0 with nvm and vanilla node symlinked to /usr/local/ but the install script still hangs on the mmmagic dependency.  I might try to grab a fresh SD card tomorrow to load Jessie, and see if I can get gcc and node 4 to work.  I didn't see anywhere in the openframe install script that directly references v4.  Am I missing something?

jmwohl, 2016-04-10:
Ah, sorry man. That's for trying it though! I don't think you'll have any issues on a fresh install with Jessie, it's what we're using and we haven't had problems.

The install script [uses nvm to install node 4.3](https://github.com/OpenframeProject/openframeproject.github.io/blob/master/install.sh#L73-L75). We may move to 5.x at some point, but for now since 4.3 is the LTS version, we're sticking with it.

indefinit, 2016-04-11:
Ok cool, I'm able to install fine on raspbian jessie.  Is it possible to add a note to the quick start doc regarding the operating system / compiler requirements?

closed 2016-04-11 by indefinit

jmwohl, 2016-04-11:
@indefinit - Cool, added a note to the user guide doc. Thanks for helping to test this!