OpenframeProject/Openframe #16: issue pushing artwork to openframe

pauloguerraf, 2016-06-02:
hi guys, 

i have installed openframe on a raspberry pi 2. everything works perfect until I try to update the artwork displayed on the frame from the website. Nothing happens when I choose a new artwork from my collection unless I hit ctrl+C and restart openframe. Only then the new artwork is displayed. 
Is this an issue? or is this the way it should always work? any advice on how to get this working would be greatly appreciated. thank you!

paulo

jmwohl, 2016-06-03:
Hi Paulo —

Yeah, that's definitely not the way it should work. It sounds like the websocket connection isn't being kept alive between your frame and the server after the frame authenticates. It's a bit hard to diagnose these issues, but could be due to a weak wifi signal or underpowered wifi dongle, or possibly the network you're on is blocking the port used for the websocket connection.

You can print debug log by adding the DEBUG env variable when starting openframe:

```
DEBUG=* openframe
```

That may give you some clues as to what's going on.

jmwohl, 2016-09-24:
Closing until we have further info  — couldn't reproduce on a fresh install.

closed 2016-09-24 by jmwohl