OpenframeProject/Openframe #19: login-popup login failure: ofsdk.js 404

meetar, 2016-07-07:
When exporting a shader from editor.thebookofshaders.com, when I click the "LOG IN" button, nothing occurs, and the console reads:

    http://openframe.io/js/ofsdk.js Failed to load resource: the server responded with a status of 404 (Not Found) 
    login-popup:67 clicked
    login-popup:69 Uncaught ReferenceError: OF is not defined

[image]

Seen on Mac OS X, Chrome 51. (The popup doesn't appear at all in FF and Safari)

jmwohl, 2016-07-08:
Thanks for reporting this — the JS sdk was moved, so likely this hasn't been working for quite a while! Should be an easy fix, will try to get to it soon.

jmwohl, 2016-07-09:
This was a problem in the [Openframe-APIServer](https://github.com/OpenframeProject/Openframe-APIServer) repo. I just pushed a patch release that fixes it — closing this issue.

Also submitted a PR that should get the popup working in Firefox. Safari requires disabling of the popup blocker.

https://github.com/OpenframeProject/Openframe-APIServer/releases/tag/v0.3.1

closed 2016-07-09 by jmwohl