OpenframeProject/Openframe #20: no start from command line

erikfx, 2016-07-08:
Hello,

I'm having issues while starting openframe from the command line. I get the error:

`Traceback (most recent call last):
File"/home/pi/.nvm/versions/node/v4.3.2/bin/glslLoader", line 24, in <module>
with open (SHADER_PATH) as f:
10Error: No such file or directory: '$url'`

If openframe is set to autostart, everything is working. Its also weird that when I quit openframe via ctrl+c there is no command line, just a black screen.

I use the raspberry 3 and I'm new to all this.

jmwohl, 2016-07-08:
Hey, thanks — I've noticed this error a few times recently as well, and haven't yet been able to track it down. It has specifically to do with shader artworks, and it doesn't occur with all of them...

Regarding autostart, the blank screen is expected (though not the best for cases like this!). In order to make sure no text output appears between or around artworks, we hide all of the terminal output. After pressing `ctrl+c` to exit openframe, you might be able to [get to a different TTY](http://unix.stackexchange.com/questions/167386/how-to-switch-between-tty-and-xorg-session/167388#167388). Otherwise you should restart the Pi, and as soon as you see the login text press 'n'. This will stop the autostart sequence.

jmwohl, 2016-07-09:
Just released a new patch version that fixes this bug — marking this issue closed.

closed 2016-07-09 by jmwohl