OpenframeProject/Openframe #23: Use HTTPS

adelevie, 2016-07-14:
Hi. This project looks really cool, and I was about to get started with it, but two things are preventing me:

1) lack of HTTPS URL to download and install arbitrary code (https://github.com/OpenframeProject/Openframe/wiki/Openframe-User-Guide#21-install-openframe)

2) Openframe.io itself doesn't use SSL. Not only is this not great for the privacy of Openframe users, but login/account pages aren't even protected by SSL, so usernames and passwords could leak out (e.g. http://openframe.io/login).

I really, really like what you all are doing with this project (I especially like the code of conduct here), and I'd like to be part of this community, but please consider using SSL. It's also never been easier to set up thanks to Let's Encrypt.

jmwohl, 2016-07-15:
Hey @adelevie —

You're definitely right on both points (which are actually really the same point :). I've used Let's Encrypt for SSL on other projects, and have been meaning to do so on the openframe server instance we're hosting at openframe.io. 

This is still a new and experimental project, and admittedly security hasn't been our biggest concern thus far. But yes — using SSL is important, and we should be doing so. We're currently working on the next iteration of the API and web app, and I'll try to prioritize setting up SSL as well.

Thanks for the input!

adelevie, 2016-07-19:
That sounds great. I've got very limited experience with setting up this kind of stuff myself, but I've been really happy using [Caddy](https://github.com/mholt/caddy) to get an A+ SSL Labs rating with minimal effort/config.

Btw, just tried out openframe on my RPi-2, and it's pretty slick! I look forward to more exploring.

javl, 2016-12-25:
@jmwohl To get back to this old issue; it seems openframe.io is hosted at DigitalOcean on a nginx server. They now offer an easy to follow tutorial explaining how to enable SSL for free using Lets Encrypt
https://www.digitalocean.com/community/tutorials/how-to-secure-nginx-with-let-s-encrypt-on-debian-8

jmwohl, 2016-12-25:
Hey @javl — Yeah, sorry it's been so long, and thanks for bumping this. Using SSL is in the plans (already implemented on the dev server), unfortunately we haven't had much time to work on things over the past few months. I'm very familiar with Let's Encrypt; I've set it up and used it in production on several projects.

So rest assured! When we get details of the next release worked out, hopefully within the next month of so, we'll also finish setting up HTTPS.

jmwohl, 2017-01-31:
Security, yay! (And just in time for the impending apocalypse.)

closed 2017-01-31 by jmwohl