OpenframeProject/Openframe #24: Up to date monitor / TV suggestions

vade, 2016-08-09:
It might make sense to have some suggestions for clean, minimal branding monitors that can be mounted with the PI in such a way to hide all but the power coord.

Obviously thin bezel displays, no large logo's, buttons on side or behind the display are important.

I'm happy to share once I arrive at some good options but it might be nice to have a wiki or a 'hardware' guide.

jedahan, 2016-08-09:
Maybe the easiest start would be adding sublist items under the HDMI monitor section, or linking to a new wiki page from https://github.com/OpenframeProject/Openframe/wiki/Openframe-User-Guide

nosarious, 2017-04-05:
Considering the movement of the documentation to http://docs.openframe.io/ and the lack of the HDMI monitor section, is this no longer a concern for people?

I was trying to find a way to get openframe running on a 3.2 inch display and found this post, but I'm suspecting openframe itself is tied to the graphics card, so TFT displays won't work...

jmwohl, 2017-04-05:
I think it's definitely still a concern, I can add what (relatively little) info we have to the docs. We just haven't had the opportunity to do much experimentation with a wide variety of displays, so don't have a lot to contribute. Would love to hear more from users about what's worked well...

We haven't spent any time trying to get a TFT working. At a workshop we taught last year one participant was trying to use a TFT, and it didn't work right off the bat, and we didn't pursue it at the time.

That said — it might be possible to write an extension for Openframe that would display artworks on a TFT? To be honest, I'm not sure what different drivers/setup/whatever is necessary to run a TFT on the Pi, I've never done it...

nosarious, 2017-04-05:
When I tried the TFT the text editor showed up on the small screen and the artwork showed up on the HDMI screen. As an interesting 'what's happening behind the scenes' concept that might be useful. 

May I suggest the documentation mention the pi hat tft displays are not supported as display devices? That would have saved me a bit of time.

For my smaller screen purposes I shall get one of the 3.5 inch hdmi screens on the market and skip the complicated driver requirements. I wanted to explore an intimate encounter with the artwork, where the screen was integrated within an object one can hold.