OpenframeProject/Openframe #25: javascript and canvas?

nosarious, 2016-11-10:
I had hoped to explore some webGL stuff, but it appears that may not be working for some of the different websites I tried. 
   https://threejs.org/examples/webgl_shader.html
I then tried some websites with canvas driven artwork.
   http://www.kevs3d.co.uk/dev/plasma/
Those don't work either.

jmwohl, 2016-11-10:
Yeah, unfortunately webGL does not yet work on Openframe — although it's possible to get webGL working in chromium by enabling the experimental GL feature on the Pi, it breaks other things. It's high on our wish list though, so hopefully future versions will support webGL by default.

Websites that use canvas *should* work, but the the Pi is not a fast machine, so they might run SUPER slow if they are performing a ton of calculations on the CPU. The examples you tried all seem pretty heavy, and I'm not surprised that they don't seem to work at all.

In answer to your question, we don't have a list anywhere of what will and won't work for websites — but that is a great idea, and we should. I'll add an issue about that on the [Openframe-Website](https://github.com/OpenframeProject/Openframe-Website) extension repo. We're slowly but surely updating docs, and this type of info should go in there.

FWIW, a simple slideshow of images and text should not be a problem at all.

nosarious, 2016-11-12:
Thank you very much for the response. I  suspected that was the reason, and I will try to experiment with the various formats I would like to explore.

jmwohl, 2017-02-01:
Closing — I've added a [related issue](https://github.com/OpenframeProject/Openframe-Website/issues/9) on the Openframe-Website repo.

closed 2017-02-01 by jmwohl

jeremydouglass, 2019-09-29:
WEBGL now works fine in a new install of Openframe with Openframe-Website. Tested with a shader link above (https://threejs.org/examples/webgl_shader.html) and with a p5.js web editor sketch in WEBGL mode.

jvolker, 2019-09-29:
I recognised that too but think it works on a Pi4 only.

jeremydouglass, 2019-09-29:
WEBGL is running here on a Pi 3b+

jvolker, 2019-09-30:
Let's continue this conversation here: https://github.com/OpenframeProject/Openframe-Website/issues/9