OpenframeProject/Openframe #27: openframe won't start. Terminal being blacked out.

killa-kyle, 2016-12-13:
I'm on a fresh install of Raspbian Jessie on a RPi3. Once I install open frame I get these three errors on startup:

manpath: can't set the locale; make sure $LC_* and $LANG are correct
openfsetterm: terminal xterm-256color does not support --blank
setterm: cannot (un)set powersave mode: Inappropriate ioctl for device

and then the terminal is blacked out.  nothing (that I can see) happens when I type the 'openframe' command

[Someone else is having the same problem it looks like](http://raspberrypi.stackexchange.com/questions/48094/setterm-terminal-xterm-256color-does-not-support-blank)

jmwohl, 2016-12-13:
It seems like you might be trying to launch openframe from xterm within the
desktop environment? If so, please exit the desktop environment and run
openframe directly from the terminal. Besides potential issues like these,
the desktop environment consumes resources which are of precious value on
these tiny computers.

killa-kyle, 2016-12-13:
That screen shot is me sshed into the pi. The desktop environment launches even though I chose to run openframe on boot and the pi is also ignoring holding the "N" key down on boot. I'll look into disabling the Linux desktop

jmwohl, 2016-12-13:
Ok — if you run *sudo raspi-config* from xterm, there is a setting in there
to boot to the terminal rather than to the desktop env.

jejip, 2018-03-31:
I have the same problem, which makes the raspberry uncontrollable either through ssl or when connecting a keyboard. It always boots into terminal, so it's not related to that. Might it be related to the setterm command in openframe's autoboot.sh?

The strange thing is, even when I exit openframe on the Pi (ctl+c), the terminal is still blacked out. I can start up the GUI, but when I open up the terminal, openframe will automatically start. Similarly, when I SSH into the Pi, openframe will autostart too. 

Pressing 'n' during startup doesn't work either, openframe always starts whenever the terminal opens.

Any idea how to stop openframe from always starting and messing with the setterm commands to hide the terminal?

jvolker, 2019-09-21:
I solved this problem and described it in the forum: https://groups.google.com/d/msg/openframeio/X1BapUmEchs/QPE9UUzjAgAJ

> I've been able to fix the SSH login issue by disabling Openframe autostart. This can be done by resetting the frame via `openframe -r` or commenting out the following line in `~/.bashrc`: `source ~/.openframe/autoboot.sh` 
> 
> It seems like those setterm errors have nothing to do with this being unable to login via SSH. Instead, it seems like the Openframe starting routine (including `setterm --foreground black --background black --cursor off --clear all`) is initiated every time you login via SSH.
> 
> On the hand commenting out `setterm -powersave off -blank 0` in `~/.bashrc` removes those setterm errors but I'm not sure if they caused any problem anyway.

jvolker, 2019-09-21:
We should add a note to the developer docs.

jvolker, 2019-09-21:
@jejip Both, disabling autostart as described above, as well as pressing 'n' as soon as the welcome screen appeared worked for me. Hope this helps.

jvolker, 2019-09-23:
Fixed by #45

closed 2019-09-28 by jmwohl

ericfriedman, 2021-04-13:
What if you cannot run terminal commands because it just hangs after the 

`terminal stopped working raspberry pi setterm: terminal xterm-256color does not support --blank openframe`

message?

PR #45 (2019-09-28): Fixes black window on SSH login
Fixes #42 
Fixes #27