OpenframeProject/Openframe #28: Only displaying shaders Raspi 2

dwarrenku, 2016-12-31:
It seems that my install of Openframe is operating fine, but it can only display shaders. When I attempt to load an image, jpg or other, it only displays a blank grey screen. I have installed it on a raspberry pi 2, so I'm wondering if I have an issue with memory?

Below is an image of the console. It loads shaders fine, and when I try and display a jpg it does nothing but show a blank grey screen.
https://goo.gl/photos/7JtGxaQnycXtwjXQ9

jmwohl, 2016-12-31:
Hmm — looking at the log, it does seem like a memory issue. We have run Openframe without issues on both RPi 2s and 1s though, so it's possible to get it running... Is it a fresh install of Openframe? Are there other processes running (and consuming memory) on the Pi?

Did you trying to push images from the stream, or images that you've added? Does it fail for all images? If you haven't, try displaying a small (say 100x100 pixel) image to see if that loads. 

Another thing you could try is upping the gpu memory. In `/boot/config.txt` (or via `raspi-config`), try setting `gpu_mem=256` or even `gpu_mem=512`.

dwarrenku, 2016-12-31:
Thank you, it ended up being the memory allocation to the gpu. I increased it to 256 and it worked fine.

jmwohl, 2016-12-31:
Great!

closed 2016-12-31 by jmwohl