OpenframeProject/Openframe #31: Blank screen - server can't be found

mattfelsen, 2017-02-02:
Hi! I had my openframe off for the last couple weeks since I grabbed the rPi USB cord for something else. I plugged it back in the other day, I think after the update was rolled out, and only got a black screen instead of any graphics output.

I ran the update just now, and am still getting the same thing.

I ssh'd into the rPi and when I try to run `openframe` I get this output in the console:
`[o]   ERROR: The server is not available.`

I'm running on an rPi 1 model B, which had some issues previously loading images or video, but worked fine for shaders.

jmwohl, 2017-02-03:
Sorry you're having issues getting it running again!

That error suggests it's trying to hit the old server — with the new release, the API server is at https://api.openframe.io. This should get updated automatically when you update openframe... Just to be clear, did you update openframe by running the `install.sh` script again?  

```terminal
$ bash -c "$(curl https://openframe.io/install.sh)"
$ source ~/.bashrc
```

If not, try that. If that's what you did and it still doesn't work, it's possible something is broken for the rPi 1 or with the install script.

You can check which version of `openframe` you're running using `npm list -g openframe`. It should be `0.4.7`.

mattfelsen, 2017-02-03:
Hi Jonathan. Yep, that's how I upgraded...just for kicks, I tried upgrading again and am attaching the output below. Of note, npm reports the installed version is still 0.3.11.

```
pi@raspberrypi:~ $ bash -c "$(curl https://openframe.io/install.sh)"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  3561  100  3561    0     0   7197      0 --:--:-- --:--:-- --:--:--  7223
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 11329  100 11329    0     0  57742      0 --:--:-- --:--:-- --:--:-- 58097
=> nvm is already installed in /home/pi/.nvm, trying to update using git
=> => Compressing and cleaning up git repository
Counting objects: 6072, done.
Compressing objects: 100% (6034/6034), done.
Writing objects: 100% (6072/6072), done.
Total 6072 (delta 4011), reused 2061 (delta 0)

=> nvm source string already in /home/pi/.bashrc
=> bash_completion source string already in /home/pi/.bashrc
=> You currently have modules installed globally with `npm`. These will no
=> longer be linked to the active version of Node when you install a new node
=> with `nvm`; and they may (depending on how you construct your `$PATH`)
=> override the binaries of modules installed with `nvm`:

/home/pi/.nvm/versions/node/v4.3.2/lib
├── openframe@0.3.11
├── openframe-glslviewer@0.2.5
├── openframe-image@0.2.0
├── openframe-video@0.2.1
└── openframe-website@0.2.7
=> If you wish to uninstall them at a later point (or re-install them under your
=> `nvm` Nodes), you can remove them from the system Node as follows:

     $ nvm use system
     $ npm uninstall -g a_module

=> Close and reopen your terminal to start using nvm or run the following to use it now:

export NVM_DIR="/home/pi/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && \. "$NVM_DIR/nvm.sh"  # This loads nvm
source nvm
install node
v6.9.5 is already installed.

Now using node v6.9.5 (npm v3.10.10)
install server utils
Reading package lists... Done
Building dependency tree       
Reading state information... Done
x11-xserver-utils is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 132 not upgraded.
install openframe
/home/pi/.nvm/versions/node/v6.9.5/bin/openframe -> /home/pi/.nvm/versions/node/v6.9.5/lib/node_modules/openframe/bin/cli.js

> openframe@0.4.7 install /home/pi/.nvm/versions/node/v6.9.5/lib/node_modules/openframe
> ./install.sh

Openframe -- install.sh
Update default extensions
non armv7l
/home/pi/.nvm/versions/node/v6.9.5/lib
└── openframe@0.4.7 

install default format extensions

> openframe-image@0.2.0 uninstall /home/pi/.nvm/versions/node/v6.9.5/lib/node_modules/openframe-image
> ./uninstall.sh

Openframe Image -- uninstall.sh
non armv7l
[... 77 lines cut ...]
input not recognised, must be yes or no
have you rotated your screen from default (normally landscape)? (y/n)
input not recognised, must be yes or no
have you rotated your screen from default (normally landscape)? (y/n)
n

If you have changed your display rotation, you must restart the Pi by typing: sudo reboot

If not, you must run the following command: source ~/.bashrc

After restarting or reloading .bashrc, you can launch the frame by just typing:

openframe

pi@raspberrypi:~ $ source ~/.bashrc
setterm: terminal xterm-256color does not support --blank
setterm: cannot (un)set powersave mode: Inappropriate ioctl for device

[1]+  Done                    nohup openframe > /dev/null 2>&1

npm list -g openframe

/home/pi/.nvm/versions/node/v4.3.2/lib
└── openframe@0.3.11 
```

jmwohl, 2017-02-03:
Thanks, that's helpful! So it looks like it did install the updated version, but sourcing your .bashrc didn't switch you to the new version of node. I'll have to look into why this isn't working.

For the moment, can you try manually switching to node v6.9?

$ nvm install 6.9

Then check the version of openframe again. If it's the latest version, try running openframe.

I'm on the road today but will try to replicate this issue tomorrow and get it fixed.

mattfelsen, 2017-02-03:
Sure thing, I'll check it out once I'm back at home. It may need an `nvm use 6.9` somewhere in there too. No rush on any of this btw, I appreciate the help (and the project!) 😸

jmwohl, 2017-02-05:
Ok — I made [a change](https://github.com/OpenframeProject/openframeproject.github.io/blob/master/install.sh#L76) to the `install.sh` that should fix this going forward, I think. As it was, node 6.9 was never being set to the default for the system, so if you'd previously installed openframe (or a different version of node via nvm), it would fail.

This _might_ still require a reboot rather than just sourcing the `.bashrc`, I need to test that still.

mattfelsen, 2017-02-07:
I tried giving to revist this earlier, and could neither access the pi over the network, nor see what was going on on-screen since the screen went black (I don't have a non-bluetooth kb/mouse handy so I couldn't poke around via the console). I ended up just imaging a new card and now I'm up & running (and enjoying your Moire Variation 😄 ). I'll keep an eye on this for future upgrades though...thanks for your help thus far! I still have the old card in case there's any testing you'd want to see...

jmwohl, 2017-02-07:
Thanks Matt! Glad you're back up, sorry the upgrade didn't go smoothly — clearly that's something we need to work out for future releases (I've been looking at ways to do automatic updates :).

I still plan to do some tests, installing the previous version and then upgrading... I want to make sure that it works for most users, and update the docs if a reboot is required.

closed 2017-02-07 by mattfelsen