OpenframeProject/Openframe #33: Make it possible to run on the EO hardware

iangilman, 2017-07-04:
Now that Electric Objects has discontinued their hardware business, there are likely to be people with the existing devices who are interested in doing new things with them. Also, while Giphy is  supporting the EO service, we shouldn't assume they will do so forever. Also, it would be neat! :-)

@dariusk wrote up an article a while ago about hacking the EO1:

https://feeltrain.com/blog/hack-your-eo1/

… which may have some relevant information. Also, he's indicated interest in working on this feature, but I'll let him speak for himself on that front!

I probably can't be of much help, but consider me an interested party!

jvolker, 2019-09-21:
I'm not sure about the effort to port Openframe to Android, nor do I know much about the Hardware of EO. But does it come with HDMI input? If not directly accessible maybe you could unscrew the back cover and find it there. In that case, it's probably easiest to use a Raspberry Pi with Openframe and connect it to the screen.

iangilman, 2019-09-23:
Good point. There isn't an HDMI input outside the case, but I haven't opened it up to look. I probably won't, either, as long as it still working as is. Anyway, thanks for the idea!

jvolker, 2020-04-01:
I've moved the content of this conversation to the new forum where it may be continued: https://openframe.discourse.group/t/make-it-possible-to-run-on-the-eo-electric-objects-hardware/88

closed 2020-04-01 by jvolker

itsthisjustin, 2023-03-31:
@jvolker any chance you can put your linux tutorial up again? I'm trying to get some use out of my 2 eo1 frames and unfortunately seemed to have bricked one by uninstalling the electric objects apk.

whatrocks, 2023-03-31:
> @jvolker any chance you can put your linux tutorial up again? I'm trying to get some use out of my 2 eo1 frames and unfortunately seemed to have bricked one by uninstalling the electric objects apk.

Here's my guide using I think the same or very similar approach: https://charlieharrington.com/run-linux-on-electric-objects-eo1-wall-computer/

itsthisjustin, 2023-03-31:
> > @jvolker any chance you can put your linux tutorial up again? I'm trying to get some use out of my 2 eo1 frames and unfortunately seemed to have bricked one by uninstalling the electric objects apk.
> 
> Here's my guide using I think the same or very similar approach: [charlieharrington.com/run-linux-on-electric-objects-eo1-wall-computer](https://charlieharrington.com/run-linux-on-electric-objects-eo1-wall-computer/)

Ha yeah have this pulled up too. Was missing maybe the obvious steps of how to flash the image to the device but I think I'm on a good path now after some googling of usb linux flashing tutorials. On a side note, I luckily had 2 of these and the one that I didn't mess with you can simply open a browser with WIN+B and install any apks you want. Unfortunately on the other one I deleted the base APKs it came with in hopes it would just boot to Android OS and now it's stuck at the loading screen infinitely.

jhaenchen, 2024-12-21:
I did not have any luck with Win+B, but I was able to connect a microusb to my raspberrypi running Ubuntu (that had the appropriate drivers installed) and then use ADB to install a new launcher and set that as the home app. Then on reboot it launches to the launcher. The board is an Amlogic S805 with Android installed. Actually it was quite fun to open this thing up.