OpenframeProject/Openframe #34: glslViewer: error while loading shared libraries: libGLESv2.so

r21nomi, 2017-09-09:
I installed openframe to RPi3 (Raspbian is installed by Noobs) and tried to start it but an error happened.

```powershell
pi@raspberrypi:~ $ openframe

glslViewer /tmp/shader.frag
glslViewer: error while loading shared libraries: libGLESv2.so: cannot open shared object file: No such file or directory
```

No artwork is shown.

FYI, I also installed [LCD driver](https://github.com/goodtft/LCD-show) into RPi3.

znewsome, 2017-09-09:
I am having the same issue - can't display anything via openframe due to "lslViewer: error while loading shared libraries: libGLESv2.so: cannot open shared object file: No such file or directory"

I am running Raspbian Stretch Lite, but had same error when installing Raspbian via NOOBS per the openframe documentation.  I have read that in Raspbian Stretch the library was renamed to  "libbrcmGLESv2.so" and not "libGLESv2.so" as in Jessie. Does the openframe code need to be updated to reflect this change?

jmwohl, 2017-09-09:
Hmm. I don't think we've run into this previously. Openframe uses [glslViewer](https://github.com/patriciogonzalezvivo/glslViewer) to display shader artworks — that's where the error is originating.

So most likely something in glslViewer needs to get updated. But it's possible that the way we're installing glslViewer within Openframe is outdated, which I can look into.

jeffcrouse, 2017-09-18:
I am also having this issue on a freshly installed RPi 3. I also tried installing glslviewer with apt-get but no dice. The strange thing is that even images cause this error.

jmwohl, 2017-09-18:
Ok, I talked to the author of glslViewer, evidently it has to do with the newest version of Raspbian in which they changed the location or name of that file. He needs to create a new .deb for the glslViewer package.

That said, we should be able to fix this by switching back to our old method of installing glslViewer via Openframe which was to build it source. I'll try to find time to make that temporary change so that fresh installs work.

Regarding images, the image extension actually uses glslViewer to render — so it makes sense that images are broken too. Dang.

jeffcrouse, 2017-10-09:
awesome, thanks! Could you please update here when you get a chance to make the switch?

telemetry, 2017-11-14:
@jmwohl could you recommend a previous version of Raspbian to install that would be a temporary fix? thanks for your time , big fan of the project

jmwohl, 2017-11-14:
@telemetry I'd try this Raspbian image, the last version of Jessie before they switched to Stretch: 

https://downloads.raspberrypi.org/raspbian/images/raspbian-2017-07-05/

I haven't tested it, but I'm _pretty_ sure it should work.

jmwohl, 2017-12-01:
Update on this — I still haven't had a chance to implement a fix (soon, I hope!), but it might work to [build glslViewer manually](https://github.com/patriciogonzalezvivo/glslViewer#installing-on-raspberry-pi) on the Pi after installing openframe. Not tested, but makes sense that this should work.

closed 2017-12-02 by jmwohl

jmwohl, 2017-12-02:
Happy to say this is resolved! Sorry it took so long, finally managed to find a few minutes free to take care of it.

jejip, 2017-12-07:
I still get the same error after updating openframe to the last version. Should any other steps be taken to get the fix?

jmwohl, 2017-12-07:
Try removing the current version of glslViewer, then reinstall just the openframe-glslviewer extension. 

Type 'which glslViewer' to find the existing bin, delete it. Then type 'openframe -i openframe-glslviewer'.

jmwohl, 2017-12-07:
@jejip when I get a chance I'll update that extension again to do this automatically in order to cover this case

jejip, 2017-12-07:
That works to eliminate the _error while loading_ message! 
Only now I get _warning while compiling vertex shader_ and _fragment shader_ when I load up shaders or images... Is that related or a different topic?

jmwohl, 2017-12-07:
Probably unrelated, but what's the error? Try increasing the graphics memory in raspi-config to like 256, if you haven't.

jejip, 2017-12-07:
That did the trick! Working like a charm now! 🙌

jmwohl, 2017-12-07:
Rad. Happy to help, glad it's working!