OpenframeProject/Openframe #35: TypeError: pm.init is not a function

mattun, 2017-09-11:
now i tried DEBUG=* npm test .

TypeError: pm.init is not a function
at Context.<anonymous> (test/plugin-manager.spec.js:18:8)

I think plugin-manager version is not correct?
(before one problem, apeared message can not find plugin-manager, so I changed plugin-manager.spec.js header  pm = require('../plugin-manager'); to pm = require('plugin-manager');
and npm install plugin-manager --save)

jmwohl, 2017-09-11:
Yeah sorry — the tests are neither complete nor up to date. Would be great to see someone contribute to the test suite :D...

To clarify one thing, the 'plugin-manager' module is an internal module, and was renamed [extension-manager](https://github.com/OpenframeProject/Openframe/blob/master/src/extension-manager.js). It has no relation to [this plugin-manager module in npm](https://www.npmjs.com/package/plugin-manager).

mattun, 2017-09-13:
Thanks for reply , I try install again

PR #38 (2017-12-15): extension-manager: fixed tests & variable names
This PR fixes #35 and is a response to the "work needed" tag on that issue. It is a continuation of some previous refactoring work which renamed plugins to extensions. There are a few changes to variable names and documentation. Some deprecated functions are now removed. More importantly, the related tests now pass.
closed 2017-12-15 by jmwohl