OpenframeProject/Openframe #39: Project status

jvolker, 2019-08-16:
Is this project still maintained? Thanks.

jmwohl, 2019-08-16:
Sort of? It's not under active development though. We have not done much with it in the past year, but still plan to fix bugs as time allows, and certainly will review and accept worthwhile PRs 😀.

jvolker, 2019-08-18:
Thanks, that's great to read! 

I could certainly imagine to help out with realising some of the features that I suggested in the Google group but don't know where to start as I don't know enough about the whole framework yet. It would be great if you could roughly point out what it takes to realise the features. Like in what extension/module the features would have to be implemented, explain the steps that need to be done, how to set up a dev environment and estimate the effort. 

In case I feel confident enough with it I would be happy to help out.

jvolker, 2019-09-02:
Thanks for all of the replies on the Google group. I won't have much time within the next weeks but hope to pick up some of it later on.

closed 2019-09-22 by jvolker