OpenframeProject/Openframe #40: Errors when loading images and shaders

jvolker, 2019-08-16:
I didn't manage to display anything on the screen yet. I'm getting the following errors.

When loading an image:
[image]

When loading a shader:
[image]

How can I fix this?

Thanks.

jvolker, 2019-08-16:
I've collected a bit more info.

I'm running the latest version of Raspbian Buster with desktop 2019-07-10 on a Raspberry Pi 3 Model B V1.2.

I've switched from an Elcrow 7 inch display that used a custom resolution (1024x600) to a standard screen to make sure this was not the problem.  But I'm still unable to display anything.

With images:
[image]

With shaders:
[image]

With videos:
[image]

With websites:
[image]

Any hints are greatly appreciated. Thanks.

jvolker, 2019-08-17:
The issues Adrian Goris is facing in the Google group seem to be related: https://groups.google.com/forum/#!topic/openframeio/ulD66rL-MUs

He seems to be working on an up to date SD image with open frame preinstalled.

jvolker, 2019-08-17:
I downgraded to the latest version of Raspbian Jessie (version "2017-07-05" according to https://en.wikipedia.org/wiki/Raspbian) from http://downloads.raspberrypi.org/raspbian/images/ as recommended on the Openframe website.

After startup I ran `sudo apt-get update` and `sudo apt-get dist-upgrade` before installing Openframe.

Shaders are working now but non of the other formats.

adriangoris, 2019-08-18:
Yes. I received the same error when I followed the docs. 

I was able to resolve the ibGLESv2.so error by compiling glslviewer from source. 
I now have an issue with GLFW and x11. 
`GLFW error 0x10008: X11: Failed to open display :0`

Which I believe has to do with xauth, but I haven't had time to dig into it yet.

jmwohl, 2019-08-18:
Thanks for working on this guys! An update — I think I found the problem. 

I installed a fresh image of the latest Raspbian and tried installing `glslViewer` via `apt-get`, and to my surprise it worked without issue... no glsl errors. The two extensions which rely on glslViewer are the image and shader formats, and looking at those I realized that the _image_ extension was installing a pre-compiled executable of glslViewer rather than using apt-get!

I've updated that extension install script and released a new version to NPM. I just did a fresh install of `openframe`, and it seems to have worked as expected... shaders and images are working fine. 

One other tip: make sure you're increasing the gpu memory to like 512 (via raspi-config or directly in config.txt), otherwise some larger images will fail.

jmwohl, 2019-08-18:
To be clear, the above resolves the issue with glslViewer for images and shaders. Videos with a valid URL and encoding (h264, e.g.) should work fine, just tested that with a known video without a problem on the fresh openframe install.

There _is_ still an issue with the website extension.

jvolker, 2019-08-18:
Thanks, guys!

I've just done an update of Openframe, still running Jessie. Images, videos and shaders are fine now.

I tested with this video: http://www.html5videoplayer.net/videos/toystory.mp4 

Maybe video was working all along and I was just confused by content in the stream that was not available anymore. Seems like the Openframe public stream should be cleaned up. A lot of content there seems to be unavailable. Could the website/server check this and mark it as available and maybe notify users that their content went offline?

> One other tip: make sure you're increasing the gpu memory to like 512 (via raspi-config or directly in config.txt), otherwise some larger images will fail.

Thanks, would be good to have that in the setup guide.

adriangoris, 2019-08-18:
Thank you, Johnathan.

I completely overlooked which glslViewer was being called and assumed it
was the local systems binary.

Perhaps this is something we can explicitly define in the the config file
in the future.

Cheers,
Adrian

jvolker, 2019-09-02:
> There is still an issue with the website extension.

I've posted [a solution](https://github.com/OpenframeProject/Openframe-Website/issues/10#issuecomment-527199025) that worked for me.

jvolker, 2019-09-06:
I'm waiting with closing this issue until the website extension has been fixed.

jvolker, 2019-09-29:
The website extension has been fixed with a new release. https://github.com/OpenframeProject/Openframe-Website/releases/tag/v0.2.8

In case you still have trouble setting up Openframe check out this guide with a ready made SD image: https://gist.github.com/jvolker/96a52b05459316643f8e110ff46b8e32

Closing this now.

closed 2019-09-29 by jvolker