OpenframeProject/Openframe #49: Allow installing extensions from Github repo

jvolker, 2019-09-24:
I'm close to getting a working Processing extension ready. I would like to find an easier way ( other than the way described in the dev docs) to share it with other users for testing. Also, I'm not the owner of the original repo and can't easily push releases to the corresponding npm package.

So I came up with the idea of allowing to install extensions also from Github in this style `openframe -i github:jvolker/Openframe-Processing`

I've tried to implement it here:
https://github.com/jvolker/Openframe/tree/install-extensions-from-github

I think it should work, but I'm the following errors and don't know where they come from. On my development version of Openframe (installed at `~/Openframe`) when installing `jvolker/Openframe-Processing`extension I get:
`UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: Cannot find module 'openframe-processing'``

I've applied the same code also to the globally installed version. I don't get an error but it doesn't seem to have installed it.

I might be confused as to where the extensions get installed to. I might also have broken my Openframe installations somehow which is causing those errors.

Could someone please have a look and try it? @jmwohl any idea?
What do you think about this idea/solution in general? 

Thanks.

jvolker, 2019-09-25:
I'm new to NPM and [just learned](https://zellwk.com/blog/publish-to-npm/) that scoped NMP packages could be a solution to the problem as well. So I've pushed my version of `Openframe-Processing` to NPM: https://www.npmjs.com/package/@jeremias.volker/openframe-processing

On a fresh image I tried `openframe -i @jeremias.volker/openframe-processing`. Unfortunately, it  fails with this error:
`(node:2234) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): AssertionError: missing path`

I can imagine the code could be easily updated to work with scoped packages too. Either way, I think there needs to be an easier solution to install forked extensions than setting up the whole dev environment. It's currently hard to test on multiple devices.

jeremydouglass, 2019-09-30:
Just in case, I also attempted to add jvolker/Openframe-Processing from @jeremias.volker/openframe-processing directly via npm, rather than the openframe -i mechanism -- that didn't work either (although I am not familiar with nvm / npm, so I might have done it incorrectly).

jvolker, 2019-10-04:
I've just finished the pull request that solves this issue. Please test.  @jmwohl: Please merge and create a new release in order to ship and test extensions more easily. 

Thank you.

jvolker, 2019-10-04:
@jeremydouglass check out [this post](https://github.com/OpenframeProject/Openframe/pull/58#issuecomment-538523610) that explains how to use it.

jvolker, 2020-05-22:
Implemented in v0.4.12.

closed 2020-05-22 by jvolker