OpenframeProject/Openframe #56: NPM link problem

jvolker, 2019-09-27:
I keep getting errors while working on the [Processing extension](https://github.com/jvolker/Openframe-Processing). I think they are related to npm link.

I set up everything as described in the [dev guide](http://docs.openframe.io/dev-env-setup/#npm-link-the-extensions). The Repos are located as follows:

`~/Openframe/`
`~/Openframe-Processing/`

I recognised that there must be another step after the end of the guide as Openframe yet needs to recognise and register the extension with the server and the local config file. So from within `~/Openframe/` I did `node ./bin/cli.js -i openframe-processing`.

I was then able to work for a while. Unfortunately, at some point, I got these errors when loading content of type `openframe-processing`:

```
The artwork format is not installed: openframe-processing 
(node:6870) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 2): undefined
```

I think it happened when I installed another package like `npm install --save request` from within `~/Openframe-Processing/`. Could this be the reason in combination with npm link? If so, how could I work around it?

Once the error is there, and I try `node ./bin/cli.js -i openframe-processing` again in order to register the extension another time I get this error:
```
[o]   Installing openframe-processing extension...
(node:7167) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): SyntaxError: Unexpected token var
```

I did check the symlinks in the global node installations and the local packages `node_modules` folder. They are both active. I also checked `~/.openframe/frame.json` –  the extension property lists the `openframe-processing` extension:

```
…
"extensions": {
    "openframe-image": "*",
    "openframe-glslviewer": "*",
    "openframe-website": "*",
    "openframe-video": "*",
    "openframe-processing": "*"
  }
…
```
In the webapp the frame also show this extension as installed.

What's wrong? How can I solve this?

The only thing that helped last time was reflashing the SD card and starting from scratch.

Thanks a lot.

jmwohl, 2019-09-28:
Hey @jvolker — thanks again for your interest in working on Openframe, it's nice to see some development activity here! I'm sorry I don't have more time to be responsive and answer all of your questions quickly. 

It's been quite a while since I've had a development env set up for openframe, so unfortunately I'm not sure on this one. Installing a dep inside of the linked package shouldn't have any effect on the linking package — as you noted it's just a symlink. 

Not sure if you tried this, but perhaps try _uninstalling_ the `openframe-processing` extension via openframe cli before reinstalling it. This should remove it from the frame. Then _after_ you reinstall it, you may need to link it again. These are guesses, but might help?

I'll do my best to get a dev env set up soon so that I can test things locally as well.

jvolker, 2019-09-29:
@jmwohl No worries. And I'm very happy to be able to contribute. It's a fantastic project.

I just got it back to work. It seems the problem was unrelated to NPM. It was a syntax error in the extension I was working on. I don't get errors anymore during install or run.

I wish the above-mentioned errors would show which source file and original line number it was caused by. That would make it a lot easier. Do you have an idea on how to change that?

Thanks a lot for maintaining the project, running tests, and merging PRs! 👍

jvolker, 2019-09-30:
Opened a new issue for that question. Closing this one.

closed 2019-09-30 by jvolker