OpenframeProject/Openframe #57: Slideshow, rotator, or gallery

jeremydouglass, 2019-09-29:
How would one push a scheduled series of updates to an openframe? For example, looping over a series of images, shaders, or videos (A, B, C, A, B, C), with a change every 60 seconds?

The documentation focuses on using the web app to push manual updates. I'm wondering if there is a way that is scriptable -- either through web app API, or through a script running on the frame.

jvolker, 2019-09-29:
Have you had a look at https://github.com/jmwohl/Openframe-Slideshow/? It rotates through your collection based on a default (hardcoded) duration or allows you to set every duration within the artwork. Though the latter should be broken atm.
Once installed it does that automatically.

It would be awesome if this extension would become a default extension at some point with settings to adjust and a play/pause button within the web app.

jvolker, 2020-04-30:
There is now a slideshow section in the docs: https://docs.openframe.io/#slideshow
The Javascript client might be helpful too: https://docs.openframe.io/#javascript-client

Closing this now. If you have more specific questions on this, please post in the forum. Thanks.

closed 2020-04-30 by jvolker