OpenframeProject/Openframe #62: Image/glslviewer extensions: How to switch between Pi 4 and other versions

jvolker, 2019-09-30:
Image extension and glslviewer extensions use different drivers for the graphics card depending on what version of the Pi they are installed. For example if you installed Openframe on a Pi 4 and want to switch to another Pi (using the same SD card or image) those extensions don't work anymore. Is there a way to reinstall/reconfigure the extensions to make them work on a different version of the Pi?
 
Ideally, the extensions (or Openframe) would recognise this automatically and switch to the right version.

Thanks.

jmwohl, 2019-10-21:
Does just uninstalling and re-installing the extension not work?

jvolker, 2019-10-31:
Good idea. Just tested it. But no, it doesn't. The install script stops if glslViewer is installed already:

```
# if glslViewer is already present, don't continue
if hash glslViewer 2>/dev/null; then
    exit 0
fi
```

jvolker, 2020-05-22:
I think this is pretty low priority. Closing this issue for now.

closed 2020-05-22 by jvolker