OpenframeProject/Openframe #70: Rename Repository

jvolker, 2019-10-21:
The headline of this repository currently reads "Openframe Frame Controller". I find this current name misleading as I understand this repository more as the (Raspberry Pi) client of Openframe. The web app, on the other hand, feels like the controlling part to me. 

Also, the repository URL is called "Openframe". In my opinion, this is not ideal either as this is only a small part of a larger system.

I suggest to rename it to "Openframe Client".
I also suggest changing the repository URL to "Openframe-Client".

jmwohl, 2019-10-21:
This one I don't entirely agree with — in a typical API / Client architecture, the web app is as much a 'client' as the software that runs on the frame. 

The code in this repository is in fact the software 'controlling the frame' in a very literal sense. I understand how from an app _user_ (i.e. non engineering) perspective the web app can be seen as the 'controller' for the frame, but since the code repositories are for engineers I don't think that's too important. 

All of that said, I'm not opposed to some renaming of components for clarity — but since any external application that talks to the server could be considered a client, I don't think calling this repo `Openframe-Client` adds clarity.

jvolker, 2019-10-21:
Yeah, all valid points, I think.

How about something more verbose and similar to `Openframe-Display-Controller`?