OpenframeProject/Openframe #79: Images not displaying

jzting, 2020-05-14:
I just installed on an RPi 2 and am having trouble displaying images. The image download seems to work fine but nothing displays onscreen:

```
openframe:image Artwork config:  {} +1m
// Loading /tmp/5d937477a38167076035bdeddrops--1868693889019328712-f240.jpg as the following uniform:
//    uniform sampler2D u_tex0; // loaded
//    uniform vec2 u_tex0Resolution;
Warnings while compiling vertex shader:
Compiled
Warnings while compiling fragment shader:
Compiled
Warnings while compiling vertex shader:
Compiled
Warnings while compiling fragment shader:
Compiled
```

jvolker, 2020-05-14:
What's the resolution of the image?
What's your GPU memory split?
Have you tried the [two image examples mentioned in the docs](https://docs.openframe.io/#3-displaying-artwork)?

Please make sure to read the [Getting started guide](https://docs.openframe.io/#getting-started) from the docs. If this doesn't help could you please provide the URL to the image?

jzting, 2020-05-14:
Thanks for the quick reply!

I tried with the two image examples and they both yield a black screen. I set the GPU memory split to 512MB.

The two shaders examples work, though.

jzting, 2020-05-15:
Ah, I forgot to reboot after setting the GPU split. It's working now!

closed 2020-05-15 by jzting

jvolker, 2020-05-15:
Glad it's working now. Thanks for reporting back.