OpenframeProject/Openframe #80: Error loading web site

jzting, 2020-05-15:
I'm getting an error when trying to push a web site to my frame:

  openframe:frame_controller swapArt +5ms
  openframe:frame_controller startArt +7ms
  openframe-website [Function: _replaceTokens] /home/pi/.nvm/versions/node/v6.9.5/lib/node_modules/openframe-website/scripts/.xinitrc { '$url': 'http://jzlabs.com', '$flags': '' } +0ms
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: spawn xinit ENOENT
    at exports._errnoException (util.js:1022:11)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:193:32)
    at onErrorNT (internal/child_process.js:359:16)
    at _combinedTickCallback (internal/process/next_tick.js:74:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)

jvolker, 2020-05-19:
Looks related to #81. Closing this for now as the conversation is happening there. Feel free to reopen in case this turns out to be a separate issue.

jvolker, 2020-05-22:
Since your issue with the Processing extension had a different reason, I'm going to reopen this one. Did you make any progress on this one? 

Did you try what I recommended in https://github.com/OpenframeProject/Openframe/issues/81#issuecomment-631093166?

jzting, 2020-05-23:
This is working now, thanks!

closed 2020-05-23 by jzting

jvolker, 2020-05-23:
Great! 

Would be interesting to know what caused it. Do you have any idea?

jzting, 2020-06-04:
Sorry I missed this question! I believe this was fixed when updating the GPU memory setting.