OpenframeProject/Openframe #81: Processing sketch not working

jzting, 2020-05-16:
I tried pushing [this](https://openframe.io/artwork/5d8546bfa38167076035bd5e) Processing sketch to my frame and it gave this error:

```
 openframe:frame_controller ready +5s
  openframe:frame_controller changeArtwork { title: 'Flocking by Daniel Shiffman',
  is_public: true,
  url: 'https://www.dropbox.com/s/me3pm1lv3tx3pg9/Flocking.zip?dl=1',
  thumb_url: 'https://i.ibb.co/4PbMpt7/image.png',
  author_name: 'J',
  required_extensions: {},
  format: 'openframe-processing',
  id: '5d8546bfa38167076035bd5e',
  ownerId: '5d51eb67a38167076035bca6',
  created: '2019-09-20T21:38:07.626Z',
  modified: '2019-09-20T21:38:07.626Z',
  is_author: true } +20ms
  openframe:frame_controller details undefined undefined { title: 'Flocking by Daniel Shiffman',
  is_public: true,
  url: 'https://www.dropbox.com/s/me3pm1lv3tx3pg9/Flocking.zip?dl=1',
  thumb_url: 'https://i.ibb.co/4PbMpt7/image.png',
  author_name: 'J',
  required_extensions: {},
  format: 'openframe-processing',
  id: '5d8546bfa38167076035bd5e',
  ownerId: '5d51eb67a38167076035bca6',
  created: '2019-09-20T21:38:07.626Z',
  modified: '2019-09-20T21:38:07.626Z',
  is_author: true } { name: 'openframe-processing',
  display_name: 'Processing (pde)',
  download: true,
  start_command: [Function: start_command],
  end_command: 'pkill -f X && pkill -f java',
  xinitrcTplPath: '/home/pi/.nvm/versions/node/v6.9.5/lib/node_modules/openframe-processing/scripts/.xinitrc.tpl' } +18ms
  openframe:frame_controller download +33ms
  openframe:downloader downloading https://www.dropbox.com/s/me3pm1lv3tx3pg9/Flocking.zip?dl=1 +11ms
  openframe:pubsub pubsub client connected +1s
  openframe:downloader Artwork downloaded +2s
  openframe:frame_controller swapArt +10ms
  openframe:frame_controller startArt +7ms
  openframe:processing unzip sketch +0ms
  openframe:processing Extracted file 1 of 10 +890ms
  openframe:processing Extracted file 2 of 10 +21ms
  openframe:processing Extracted file 3 of 10 +21ms
  openframe:processing Extracted file 4 of 10 +222ms
  openframe:processing Extracted file 5 of 10 +6ms
  openframe:processing Extracted file 6 of 10 +6ms
  openframe:processing Extracted file 7 of 10 +6ms
  openframe:processing Extracted file 8 of 10 +6ms
  openframe:processing Extracted file 9 of 10 +7ms
  openframe:processing Extracted file 10 of 10 +6ms
  openframe:processing Finished extracting +16ms
  openframe:processing wrapped directory exists > move files +9ms
  openframe:processing hideBackground +113ms
  openframe:processing clone template +157ms
  openframe:processing extend tokens +108ms
  openframe:processing replace tokens +10ms
  openframe:processing [Function: _replaceTokens] /home/pi/.nvm/versions/node/v6.9.5/lib/node_modules/openframe-processing/scripts/.xinitrc { '$url': 'https://www.dropbox.com/s/me3pm1lv3tx3pg9/Flocking.zip?dl=1',
  '$id': '5d8546bfa38167076035bd5e',
  '$filepath': '/tmp/5d8546bfa38167076035bd5eFlocking.zip',
  '$filename': 'Flocking.zip',
  '$tmpSketchPath': '/tmp/Flocking',
  '$flags': '' } +7ms
  openframe:processing build command +485ms
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: spawn xinit ENOENT
    at exports._errnoException (util.js:1022:11)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:193:32)
    at onErrorNT (internal/child_process.js:359:16)
    at _combinedTickCallback (internal/process/next_tick.js:74:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)
```

I tried downloading the [zip file](https://www.dropbox.com/s/me3pm1lv3tx3pg9/Flocking.zip?dl=1) locally and it builds/runs in Processing.

jvolker, 2020-05-17:
Thanks for reporting these issues @jzting. I don't have access to my Openframe setup atm but will get back to you asap.

jzting, 2020-05-17:
Thanks for taking a look! The web page issue seems related since it has the same stack trace.

jvolker, 2020-05-17:
- Have you tried any other artwork? Anything that is working? 
- What version of Raspbian are you using and what model of the Raspberry Pi?
- What version of openframe and what version of the openframe-processing extension are you using?
- Is this a clean install or do you have any other software running on the Pi?

jzting, 2020-05-17:
* Images, videos, and shaders are working
* Buster on a Raspberry Pi 2 Model B 
* openframe@0.4.12 and openframe-processing@2.0.1
* The only other software I'm running is [Pi-hole](https://pi-hole.net/)

jzting, 2020-05-19:
Hi, any ideas based on the info above?

jvolker, 2020-05-19:
I've tried this sketch on my Pi3 Model B running Buster and it's working for me. 
It looks like xserver is not starting which is required for the Processing and Website extension.

Are you sure to run Openframe from the command line outside the Desktop environment?
Otherwise could you please try running `sudo apt-get install xserver-xorg-legacy` and see if that makes a difference?

jzting, 2020-05-20:
I found the issue: Processing wasn't installed because the 2B's architecture is armv6l. I must have missed the error message during the extension installation.

To fix this, installed Processing by following the instructions at https://pi.processing.org/download/:

`curl https://processing.org/download/install-arm.sh | sudo sh`

jzting, 2020-05-20:
I think the openframe-processing install script should check for (armv6l || armv7l) when checking if it should fetch from https://processing.org/download/install-arm.sh.

closed 2020-05-20 by jzting

jmwohl, 2020-05-20:
Thanks for following up with the solution @jzting!

> I think the openframe-processing install script should check for (armv6l || armv7l) when checking if it should fetch from https://processing.org/download/install-arm.sh.

That seems like a good idea. I believe we do that in some of the other extension install scripts.