OpenframeProject/Openframe #82: How to set aspect ratio for videos?

jzting, 2020-05-21:
I see on the [OpenFrame-Video page](https://github.com/OpenframeProject/Openframe-Video):

"Artworks can pass arguments to omxplayer using the Artwork object's config property. Probably the most useful config argument is --aspect-mode, which allows an artwork to specify how it should size itself within the display."

Where do I set this config argument?

jvolker, 2020-05-22:
Unfortunatly, there is no UI for this yet, but this section of the docs links to a guide using the API explorer: https://docs.openframe.io/#per-artwork-settings

jzting, 2020-05-23:
Got it – it would be nice if this was in the web app UI. That way I wouldn't have to manually update the artwork object's aspect ratio each time I change the artwork.

jvolker, 2020-05-23:
Yeah, this would be great and has been discussed before. Pull requests are welcome.

The aspect ratio is attached to the artwork. So it should change to whatever your artwork has been set to. Maybe this is a misunderstanding. Could you please describe your use-case a bit more? Thanks.

jzting, 2020-05-23:
I am showing videos from Instagram that are square – I'd like them to show aspect fit instead of fill.

jvolker, 2020-05-23:
So you would like a global setting that applies to all videos - is that right?
How many different videos do you have?

jzting, 2020-05-23:
Yes, a global setting would be fine. I would have maybe 10 or so videos rotating using openframe-slideshow.

jvolker, 2020-05-24:
I've opened a new issue for this here: https://github.com/OpenframeProject/Openframe-Video/issues/5

Closing this one now.

closed 2020-05-24 by jvolker