OpenframeProject/Openframe #84: POST /artwork/{id}/replace returning 401

jzting, 2020-05-23:
I've used [OpenframeUser_login](https://api.openframe.io/explorer/#!/OpenframeUser/OpenframeUser_login) to get an access token and use that token in [Artwork_replaceById_post_artwork_id_replace](https://api.openframe.io/explorer/#!/Artwork/Artwork_replaceById_post_artwork_id_replace).

I'm getting a 401 error. Is this expected?
``{
  "error": {
    "statusCode": 401,
    "name": "Error",
    "message": "Authorization Required"
  }
}``

jvolker, 2020-05-24:
Have you pressed the "Set Access Token" button first?

I'm not sure what "POST /artwork/{id}/replace" is doing exactly.  You could try "PUT /artwork/{id}" instead. I think that's what I've used in the past.

jvolker, 2020-05-24:
Or wait, what would you like to do? I was assuming you want to change the options of an artwork based on your other posts.

jzting, 2020-05-24:
In this case I'm trying to set up a script to regularly update an artwork with a new URL.

I've pressed the "Set Access Token" and also tried "PUT /artwork/{id}" and it still gives a 401. Is it working for you now?

jvolker, 2020-05-24:
I don't have the time at this moment but can try later.

In case you are using Javascript, have you seen https://github.com/OpenframeProject/Openframe-JSClient and its [documentation](https://docs.openframe.io/#javascript-client)?

jzting, 2020-05-24:
I was using the StrongLoop API Explorer but will give the JS client a try.

jvolker, 2020-05-25:
> I've pressed the "Set Access Token" and also tried "PUT /artwork/{id}" and it still gives a 401. Is it working for you now?

Just tested it. Yes, it's still working for me. Does it say "Token Set." left of the form input for the token?

Did you have luck with the JS client?

jzting, 2020-05-25:
Oops, I was copying the token into another window which seems to not work well.

When trying it in the same window, I now get a 404 and the artwork I'm trying to update disappears from the web UI.

[image]

[image]

jzting, 2020-05-25:
(I haven't had a chance to try the JS client yet)

jvolker, 2020-05-26:
I'm sorry, I'm not sure what could cause this. Which endpoint were you using when getting the last 404? I've used "PUT /artwork/{id}".

But again, I'm curious about your use case. Why would you like to update the URL? Where do you store those URLs? Could the slideshow extension be an alternative?

jzting, 2020-05-26:
I was using "PUT /artwork/{id}".

My use case is fairly niche: I post to Instagram roughly every day and want to update my frame with the latest post. I was thinking of creating a custom [Apple Shortcut](https://support.apple.com/en-za/guide/shortcuts/apd58d46713f/ios) to be able send an Instagram link to the frame.

I want to update the same artwork instead of using a slideshow because Instagram video links expire after a couple days; updating the same artwork regularly would help avoid a collection of artworks with expired links.

jvolker, 2020-05-26:
I would probably write an Openframe extension checking your Instagram account (probably using their javascript SDK) and creating a new artwork each time the video URL changes and deleting the old one.
You also need to let your frame know that the artwork has changed and reload. With an Openframe extension, you can do that and it runs on the Pi so you don't need another server/computer running the script.

jzting, 2020-05-26:
Thanks for the idea!

jvolker, 2020-05-26:
No worries. 

Would you mind closing this issue if you don't have any more questions about the API explorer? If you more question about extensions, etc. please open a new thread in the [forum](https://openframe.discourse.group/) instead. Thank you.

closed 2020-05-26 by jzting

jzting, 2020-05-26:
Thanks for the guidance here. To close the loop, I got the Apple Shortcut working using the endpoints I found by looking at the XHRs on the web app:

`POST /login` to get a token
`POST /users/create/created_artwork` to create a new artwork
`PUT /frames/{frameId}/current_artwork/{artworkId}` to set the artwork as the current artwork