OpenframeProject/Openframe #93: [o]   ERROR: The server is not available.

son1cn, 2021-10-09:
I've had Openframe running for a few months with no issue. A week or so ago it stopped initializing and I started looking into it today. When I run the open frame command, this comes up now:

[o]   ERROR: The server is not available.

I tried to reinstall via the regular install script but no avail. I then reflashed raspbian and tried the script on a newly updated OS and still no dice. Openframe 0.4.12 and node 6.9.5.

```
npm list -g openframe
/home/pi/.nvm/versions/node/v6.9.5/lib
mqq openframe@0.4.12
```

Has anyone else experienced issue recently? I checked that the pi has access and can ping api.openframe.io with no issue.

WBWBW, 2021-11-13:
Did installing from scratch solve it ?

rayber, 2022-01-28:
I tried installing from scratch and it did not work

Stefterv, 2022-02-04:
I installed a local cors proxy through npm and used that to successfully login:

`npm install no-cors-proxy -g`

`no-cors-proxy -p 8000 -t  https://api.openframe.io &`

`nano ~/.openframe/.ofrc`

```
{
  "network": {
    "api_base": "http://localhost:8000",
    "app_base": "https://openframe.io"
  }
}
```

`DEBUG=* openframe`

Now I could login

WBWBW, 2022-02-05:
> I installed a local cors proxy through npm and used that to successfully login:
>
> `npm install no-cors-proxy -g`
>
> `no-cors-proxy -p 8000 -t https://api.openframe.io &`
>
> `nano ~/.openframe/.ofrc`
>
> ``` [...]

Awesome! Do you know how to make the proxy persist after a reboot? Thanks for the help!

rayber, 2022-02-08:
You can add it in `~/.openframe/autoboot.sh` and add the following line before the openframe line

```
nohup no-cors-proxy -p 8000 -t https://api.openframe.io & >/dev/null 2>&1 &
```

bimalnair, 2022-02-14:
Hi I followed the above instructions and I still get

ERROR : The server is not available

Openframe: cli Can't read from server. It may not have appropriate access-control-origin settings.

What can be done to solve this?

jmwohl, 2022-02-14:
The cause of this issue is the `swagger-client` dependency. For whatever reason, using this client to connect to the server no longer works. I haven't had a chance to look into it in detail, but one solution would be to replace the use of this package with a lower-level HTTP request package and make the API requests more manually, or perhaps swap in the [openframe-jsclient](https://github.com/OpenframeProject/Openframe-JSClient) and use that for API requests. 

Contributions very welcome!!

RasmusNielsen, 2022-09-10:
Excited for this project. Hope this can be kept alive.

hannsQ, 2023-01-19:
hello there :) ... seems that i ran into this problem also, fresh install of a rasberry pi 3, follwed the instructions to install openframe, bit i get this Error: The server is not available.
this is the debug output:

setterm: terminal xterm does not support --blank
setterm: cannot (un)set powersave mode: Unpassender IOCTL (I/O-Control) für das Gerät
piop@raspberrypi:~ $ DEBUG=* openframe
  openframe:config load +0ms
  openframe:frame load +16ms
  openframe:frame inflateStateFromFile +6ms
  openframe:user load +4ms
  superagent GET https://api.openframe.io/explorer/swagger.json +107ms

[o]   ERROR: The server is not available.

  openframe:cli Can't read from server.  It may not have the appropriate access-control-origin settings. +461ms

is there any solution for this ? 
thanks !!

jmwohl, 2023-01-19:
Hi @hannsQ — I haven't followed up on this, but I'm fairly certain the solution is replace the `swagger-client` dependency with simple fetch requests to the API (or maybe to use a different or updated swagger client).

hannsQ, 2023-01-20:
thank you jmwohl for your answer, i just dont know how to do that - where can i replace the swagger-client and with which kind of commad do i make this fetch request to the api?

Thank you !!

during installation of openframe following error occured:

E: Unable to locate package glslviewer
npm ERR! Linux 5.15.84-v7+
npm ERR! argv "/home/pi/.nvm/versions/node/v6.9.5/bin/node" "/home/pi/.nvm/versi                                                    ons/node/v6.9.5/bin/npm" "install" "-g" "openframe-image@latest"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! openframe-image@0.2.3 install: `./install.sh`
npm ERR! Exit status 100
npm ERR!
npm ERR! Failed at the openframe-image@0.2.3 install script './install.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the openframe-image packa                                                    ge,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./install.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs openframe-image
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls openframe-image
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
/home/pi/.nvm/versions/node/v6.9.5/bin/glslLoader -> /home/pi/.nvm/versions/node                                                    /v6.9.5/lib/node_modules/openframe-glslviewer/bin/glslLoader

> openframe-glslviewer@0.2.8 install /home/pi/.nvm/versions/node/v6.9.5/lib/node                                                    _modules/openframe-glslviewer
> ./install.sh

Installing glslViewer...
./install.sh: line 36: [: ==: unary operator expected
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package glslviewer
npm ERR! Linux 5.15.84-v7+
npm ERR! argv "/home/pi/.nvm/versions/node/v6.9.5/bin/node" "/home/pi/.nvm/versi                                                    ons/node/v6.9.5/bin/npm" "install" "-g" "openframe-glslviewer@latest"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! openframe-glslviewer@0.2.8 install: `./install.sh`
npm ERR! Exit status 100
npm ERR!
npm ERR! Failed at the openframe-glslviewer@0.2.8 install script './install.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the openframe-glslviewer                                                     package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./install.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs openframe-glslviewer
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls openframe-glslviewer
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
 ...
also starting npm with debug flag give this result:

pi@rasp:~ $ DEBUG=* npm start
npm ERR! Linux 5.15.84-v7+
npm ERR! argv "/home/pi/.nvm/versions/node/v6.9.5/bin/node" "/home/pi/.nvm/versions/node/v6.9.5/bin/npm" "start"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! path /home/pi/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open

npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
pi@rasp:~ $

thank you!!

brookstalley, 2023-03-11:
I’ve been digging into this a bit… the root cause is the API server not setting the CORS headers, at least on swagger.json. I see that the server is using loopback, and the config has cors:false under remoting, but the loopback [documentation](https://loopback.io/doc/en/lb4/Customizing-server-configuration.html) suggests there should be a cors:{origin:false} entry under the rest options. This might be a change in loopback?

I haven’t yet set up a server to test, but figured I’d share.