OpenframeProject/Openframe #97: syntax error running openframe

quinkennedy, 2024-10-30:
looks like one of the dependencies is using newer language features. 

* Raspberry PI 3 B+
* Raspbian "Debian GNU/Linux 12 (bookworm)"
* Installed via install script

```
$ openframe
/home/quinkennedy/.nvm/versions/node/v6.9.5/lib/node_modules/openframe/node_modules/tough-cookie/dist/memstore.js:30
        const result = this.idx[domain]?.[path]?.[key];
                                        ^
SyntaxError: Unexpected token .
    at Object.exports.runInThisContext (vm.js:76:16)
    at Module._compile (module.js:542:28)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/quinkennedy/.nvm/versions/node/v6.9.5/lib/node_modules/openframe/node_modules/tough-cookie/dist/cookie/index.js:6:18)
    at Module._compile (module.js:570:32)
```

quinkennedy, 2024-10-30:
anyone have a package-lock file they could provide to ensure I get the correct versions of the various dependencies?

RazZziel, 2025-12-27:
I get a similar error "SyntaxError: Unexpected token", on Debian 13 on a Raspberry Pi 5:
```
raziel@photoframe:~ $ openframe 
/home/raziel/.nvm/versions/node/v6.9.5/lib/node_modules/openframe/node_modules/tough-cookie/dist/index.cjs:71
  options = { ...defaultGetPublicSuffixOptions, ...options };
              ^^^
SyntaxError: Unexpected token ...
    at Object.exports.runInThisContext (vm.js:76:16)
    at Module._compile (module.js:542:28)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/raziel/.nvm/versions/node/v6.9.5/lib/node_modules/openframe/node_modules/faye/src/util/cookies/node_cookies.js:3:18)
    at Module._compile (module.js:570:32)
```